kwankiu/archlinux-installer #7: offset of boot part

hbiyik, 2023-08-30:
it should be **MiB, not MB.**

0x8000 = 32768 sectors = 32768 * 512 (bytes) = 16 * 1024(mebi) * 1024(kiki) = 16 x 1048576 bytes
(Gnu is picky about units...)

tested with my own sd card did not work. and noticed the unit was  wrong. Did not test with this change as well.

I am curious how it worked with 16MB offset, it should have not worked

https://www.gnu.org/software/parted/manual/html_node/unit.html

kwankiu, 2023-08-30:
I thought that doesnt matter as I see from Radxa and Armbian's image, I see different boot partition size (showing 15MB and 16.xMB). I will add this change to dev and try 

sudo parted $drive mkpart primary fat32 16MiB 500MiB

hbiyik, 2023-08-30:
ok, now i understand, armbian's bootloader is a modded version, i am on default radxa/rockchip bootloader. And i can confirm it only works with offset 16 MiB offsets with my sdcard. Nvme should be the same.

hbiyik, 2023-08-30:
when it is 16MB it does not work in other words

hbiyik, 2023-08-30:
update: the script works i guess, now it is installing panfork-mesa, if it is up to here then it will work. Thanks again for the script...

kwankiu, 2023-08-30:
Ok changes merged on commit #cff82e9

closed 2023-08-30 by kwankiu