kwankiu/archlinux-installer #8: Disable Wayland and use X by default

hbiyik, 2023-08-31:
wayland in the vendor kernel is really problematic, it works on some kernel versions and then breaks again after udpate, and sometimes  on the same kernel updating the firmware breaks it again. So i would suggest to disable it and use X by default. This would fix all those stuck cases.

I have a feeling it seems to be realted to DRI but i dont really care, i hope some wayland expert in future fixes it..

so for gnome's case
/etc/gdm/custom.conf

reuqires such an entry, by default it is commented out
```
[daemon]
# Uncomment the line below to force the login screen to use Xorg
WaylandEnable=false
```

kwankiu, 2023-08-31:
What stuck cases do you experience? Since I was using Wayland with Vendor Kernel for a while. I would say except for the random freeze on inactivity I am not sure whether this is related to Wayland or not. Wayland environment is at least usable. The graphics on X is also glitchy.

hbiyik, 2023-08-31:
easiest to duplciate is run gnome-calendar. it wont show up.
try to shutdown sbc, it will shutdown every process but sbc wont shutdown.

kwankiu, 2023-09-01:
I think I can do it like that:

options=("Gnome (X only)" "Gnome (Wayland Experimental)" "KDE Plasma" "Budgie")

Gnome (X only) will install a full gnome desktop with Wayland Disabled
Gnome (Wayland Experimental) will install a minimal gnome desktop which will not install software like epiphany, gnome-calendar, etc.

I am not sure about other dm and de. I think Budgie is also a bit buggy.

hbiyik, 2023-09-05:
i tested and it works when x gnome is selected

kwankiu, 2023-09-05:
I actually didn’t close this issue because i dont find X really more stable. In my case, I see that terminal app is not so responsive on X, i am not sure whether this is X or the new kernel, as i see that the performance governor cant be set in the previous way. So i have to test. I remember also seeing other apps not launching or glitches.

closed 2023-10-17 by kwankiu