kwankiu/archlinux-installer #9: Dkms packages

hbiyik, 2023-08-31:
Radxa kernel is disabling following drivers ftom the kernel and will build them via dkms.

R8125: onboard ethernet driver
Rtl8852bu/be: wlan bt driver
and some more

https://github.com/radxa-repo/bsp/blob/main/linux/rockchip/kconfig.conf

If you are not in detail about dkms, they are source packets that contain kernel modules which are compiled dynamically whenever a new kernel is installed.

This will impact to the flow of installer script because you need network connection to move forward, but it will only be available whenever dkms packages are built.

So the flow would be:
Install vendor kernel
Install dkms drivers (has depenedencies like dkms, make, gcc and not sure if rootfs comes with them)

Reboot to kernel and update system since you would have network now.

R8125 package could be: https://archlinux.org/packages/extra/any/dkms/

8852bu/be: i will introduce new packages

kwankiu, 2023-08-31:
Thanks.

I am actually working on to update the flow of the installation script. I actually plan to provide options for both offline install and online install. Which online install will allows us to pick between kernel and packages options under “arch-rock-installer” (note that the new command is simply just “installer”). For offline install, the installation script will curl all the required pkgs from 7ji repo or other repo (if necessary) to /usr/lib/compiled-pkg, so that “arch-rock-installer” can install them offline using pacman -U

hbiyik, 2023-09-01:
here are the packages [8852bu-dkms-git](https://aur.archlinux.org/packages/8852bu-dkms-git), [8852be-dkms-git](https://aur.archlinux.org/packages/8852be-dkms-git), [r8125-dkms-git](https://aur.archlinux.org/packages/r8125-dkms-git)

kwankiu, 2023-09-01:
@hbiyik will these three packages be available on 7Ji’s archrepo? How’s the kernel works now?

kwankiu, 2023-09-02:
> here are the packages [8852bu-dkms-git](https://aur.archlinux.org/packages/8852bu-dkms-git), [8852be-dkms-git](https://aur.archlinux.org/packages/8852be-dkms-git), [r8125-dkms-git](https://aur.archlinux.org/packages/r8125-dkms-git)

How can they be installed without the Internet? I tried to run makepkg -S with them, but they generates src.tar.gz instead of pkg.tar.gz.

kwankiu, 2023-09-02:
> here are the packages [8852bu-dkms-git](https://aur.archlinux.org/packages/8852bu-dkms-git), [8852be-dkms-git](https://aur.archlinux.org/packages/8852be-dkms-git), [r8125-dkms-git](https://aur.archlinux.org/packages/r8125-dkms-git)

[8852bu-dkms-git](https://aur.archlinux.org/packages/8852bu-dkms-git), [8852be-dkms-git](https://aur.archlinux.org/packages/8852be-dkms-git), [r8125-dkms-git](https://aur.archlinux.org/packages/r8125-dkms-git)

and the new linux-radxa-rkbsp5-git 

is now on https://github.com/kwankiu/archlinux-installer-rock5/releases/tag/repo for test,

will be removed once they are on 7Ji repo

kwankiu, 2023-09-02:
@hbiyik 

I am trying to install the dkms packages without internet (we wont have internet anyways without the r8125 dkms ...). 

Pacman -U listed these packages are not installed : gc gcc guile libisl libmpc make patch dkms.
I downloaded them and pacman -U these 8 packages from alarm mirror.

at (2/2) Install dkms modules 
I am getting
Error! Bad return status for module build on kernel: 5.xxxxxxxxxxxxxxxxxx (aarch64)

Either sth to do with the kernel or still need to install some more packages

kwankiu, 2023-09-02:
Ok, installing libnsl and python solved this.

So that we need total of 10 packages : 
gc gcc guile libisl libmpc make patch dkms libnsl python

hbiyik, 2023-09-02:
ok, in case something blows up please ping me, but so far things seems to work on my rock5b. did not test 8852 though.

kwankiu, 2023-09-02:
I am still working on the installer, the new kernel change (mali and dkms) breaks it, the kernel seems installed ok now. But I am not sure 8852 gets installed or not, and all other packages (mesa, ffmpeg, mpp, etc) are broken.

kwankiu, 2023-09-02:
@hbiyik I see that the linux-radxa-rkbsp5-git is available on 7Ji's pacman repo, while ffmpeg4.4 is removed.

hbiyik, 2023-09-02:
https://github.com/7Ji/archrepo/issues/4

you can verify if the dkms module is actually installed by simply locating them `find /usr/lib/modules | grep 8852`, also `modinfo 8852bu` or the related ko name would give the installation location along with other parameters. But i cant say i understood exactly what the issue you are facing is.

hbiyik, 2023-09-02:
also i would personally suggest drop offline installer at all, arch is meant to be updated and rolled out. it would not make sense to have an image that can be installed to a device which can not update itself. If it can update, why would anyone need an offline installer. But this is only  my thought.

kwankiu, 2023-09-02:
> [7Ji/archrepo#4](https://github.com/7Ji/archrepo/issues/4)
> 
> you can verify if the dkms module is actually installed by simply locating them `find /usr/lib/modules | grep 8852`, also `modinfo 8852bu` or the related ko name would give the installation location along with other parameters. But i cant say i understood exactly what the issue you are facing is.

just ran pacman -S linux-radxa-rkbsp5-git again and checked the log. 
bc: command not found.

kwankiu, 2023-09-02:
> also i would personally suggest drop offline installer at all, arch is meant to be updated and rolled out. it would not make sense to have an image that can be installed to a device which can not [...]

I was thinking of offline install just because it is sometimes not so convenience to plug a LAN cable. And WiFi does not work on midstream tho. Moreover, not everyone is using 8852, some might be using Intel AX, or something else. But if WiFi can work from the beginning of installation, offline installer isnt much needed.

7Ji, 2023-09-02:
> [8852bu-dkms-git](https://aur.archlinux.org/packages/8852bu-dkms-git), [8852be-dkms-git](https://aur.archlinux.org/packages/8852be-dkms-git), [...]

I would hold on a sec to add those DKMS packages. They're using Radxa's source if I'm not mistaking. My repo is for multiple platforms, and most notably the platforms my following projects target:
 - https://github.com/7Ji/amlogic-s9xxx-archlinuxarm
 - https://github.com/7Ji/orangepi5-archlinuxarm

Do we have more original sources to use for the PKGBUILDs? I'm worrying about Radxa's source targeting their vendor kernel and breaking my other projects running on mainline kernels.

> I was thinking of offline install just because it is sometimes not so convenience to plug a LAN cable.

In that case why not just provide pre-built disk images and rootfs archives? Running A LOT of `pacman -U` commands is an evil source of trouble and you will mess up pacman's dep relationships as all pacakges would be considered a direct explicit install. Updating that kind of system is a nightmare. And even if you go a YOLO way by deploying the package archives directly without calling pacman, it will come back and bite you and complain about all the existing files it does not know when updating the system. Take a look at the above two of my projects, they build packages and images completely on Github Action.

hbiyik, 2023-09-02:
> I would hold on a sec to add those DKMS packages. They're using Radxa's source if I'm not mistaking. My repo is for multiple platforms, and most notably the platforms my following projects target:

i think there is a misunderstanding. Those dkms packets are not meant go to any binary or your repo since they are DKMS there is no need at all, they are source only. Not relevant but sidenote, even though they are on radxa repo, they are actually realtek original sources. I am using Radxa's repos because they are maintained with various patches so i dont have to deal with :) radxa is doing it for me and everyone. So it would work globally, but again no need to put them in binary repo they are good at AUR.

kwankiu, 2023-09-03:
> I would hold on a sec to add those DKMS packages. They're using Radxa's source if I'm not mistaking. My repo is for multiple platforms, and most notably the platforms my following projects target: [...]

The dkms may not be necessary on pacman repo for my script, as those are network driver that needs to be installed before it have network to deal with the pacman repo. What I am curious is whether the 8852 dkms would works with other kernel (opi5, armbian, midstream, mainline, etc), and maybe the Intel AX210 driver can also be packed as dkms packages?

kwankiu, 2023-09-03:
@hbiyik I found it more maintainable to build an image with the older kernel that has network out of the box so we dont have to deal with the dkms packages offline using pacman -U, additionally, manually curling packages from GitHub API is not so ideal as un-auth requests are limited to 60 per hour per IP address. 

Basically, if I leave the installation script like what it currently do in the main branch, I could remove the midstream kernel option from that and just focus on an "Online Installer" that enables options and flexibility to install latest kernel and packages from the internet.

So it might still be beneficial for @7Ji to add those dkms packages on the pacman repo. This way we dont need to make the package from AUR

hbiyik, 2023-09-03:
>> So it might still be beneficial for @7Ji to add those dkms packages on the pacman repo. This way we dont need to make the package from AUR

Then in this case the package would be made from the 7ji repo. In any case it would be compiled since it is dkms package. I do not see a difference.

kwankiu, 2023-09-03:
> > So it might still be beneficial for @7Ji to add those dkms packages on the pacman repo. This way we dont need to make the package from AUR
> 
> Then in this case the package would be made from the 7ji repo. In any case it would be compiled in the device, since it is dkms package, and it can not contain precompiled binaries. I do not see a difference.

Ok, so basically the installation is run as root, while makepkg cant run in root. To get this working, I have to modify the installer to run as user. You can see changes in dev branch.

hbiyik, 2023-09-03:
I think you should do the makepkg in the host pc, create the tar.xz, and copy this to somewhere in img, when device boots the img, first install the kerbel woth pacman, then install the .tar.xz dkms packages using pacman. Pacman at that point will compile the packages using dkms

hbiyik, 2023-09-03:
Ok let me check the script

hbiyik, 2023-09-03:
@kwankiu i understood what you mean, and checking the options. Meanwhile i fixed python and gc dependecies of dkms packages, you dont have to do this in your script. It is most likely a archlinuxarm issue, but i fixed in the pkgbuils.

hbiyik, 2023-09-03:
https://github.com/kwankiu/archlinux-installer-rock5/blob/7607b8fc1b7269e6ea4bf5105696e94e8a19bc0f/tools/arch-rock-installer#L283C5-L283C57

This needs `sudo pacman -S --overwrite \* linux-radxa-rkbsp5-git-headers`without headers you cant build the dkms

hbiyik, 2023-09-03:
So. After applying above fixed to AUR, amd arch-rock-installer it worked on my sd card.

But i think using the old kernel would cause problems in the mid or long term because this kernel will not be maintained hosted at some point and things will break.

On the otherhand having the DKMS packages offline installable requires those packages to be built in to actual packages. Now i understand this because your script is meant to be run in several places not only archlinux host distro as well, so non-arch users can not create pkgs from aur packages.

So my suggestion would be:
1) Install the new kernel
2) download all dkms dependencies and put them in the img.
3) boot to img with new kernel (no internet), compile dkms packages, (internet is available now). And move further with online script. Below is a sampkle scripts does the offline download part..

```shell
archlinuxarmurl=http://mirror.archlinuxarm.org/aarch64
sevenjiurl=https://github.com/7Ji/archrepo/releases
archlinuxarmrepos=(core extra alarm aur community sync)
# packages need to be built by dkms. they depend on dkms + python + bc
dkmspkgs=(8852bu-dkms-git 8852be-dkms-git r8125-dkms-git)
# below list is manually created by comparing the rootfs and "pactree dkms", in future the dependencies of dkms + python + bc might change
dkmsdeps=(dkms python bc gcc libmpc libisl make guile gc patch gbm libnsl)
pkgregex="-[0-9][.0-9a-z-]*"

# archlinux arm does not come with dkms preinstalled. Download dkms and all it deps
for pkg in "${dkmsdeps[@]}"
do
	echo "Searching package: $pkg"
	for repo in "${archlinuxarmrepos[@]}"
	do
		pkgname=$(curl -Lks $archlinuxarmurl/$repo | grep -oa $pkg$pkgregex | head -1)
		if [ ! -z "$pkgname" ]; then
			pkgurl=$archlinuxarmurl/$repo/$pkgname
			echo "Downloading $pkgurl"
			curl -LOJs $pkgurl
			break
		fi
	done
done

for pkg in "${dkmspkgs[@]}"
do
	echo "Searching dkms package: $pkg"
	# can not guarantee github provides below page always may be better to use api?
	pkgname=$(curl -Lks $sevenjiurl/expanded_assets/aarch64 | grep -oa $pkg$pkgregex | head -1)
	if [ ! -z "$pkgname" ]; then
		pkgurl=$sevenjiurl/download/aarch64/$pkgname
		echo "Downloading $pkgurl"
		curl -LOJs $pkgurl
		break
	fi
done
```

So you are right and it is up to you to decide which route to take. If you would like to go the offline dkms install path we should talk again with 7Ji to create 3 new packages for us.

PS: I really think the default rootfs should have included the dkms package already by default, but thats different discussion

7Ji, 2023-09-03:
Why would you need to use GitHub API at all? To get the list of packages? Why not just do it the same way as how Pacman does it? Just download 7Ji.db, it's a simple .tar file compressed with zstd. Extract it and you'll get both the package list and package file names. Append the file name to repo url and you'll get the package download URL.

kwankiu, 2023-09-03:
> https://github.com/kwankiu/archlinux-installer-rock5/blob/7607b8fc1b7269e6ea4bf5105696e94e8a19bc0f/tools/arch-rock-installer#L283C5-L283C57
> 
> This needs `sudo pacman -S --overwrite \* linux-radxa-rkbsp5-git-headers`without headers you cant build the dkms

Let me check, coz when building that package with makepkg, headed package is also generated, so I did not check whether it is needed to pacman -S the radxa-rkbsp5-git-headers package

kwankiu, 2023-09-03:
> But i think using the old kernel would cause problems in the mid or long term because this kernel will not be maintained hosted at some point and things will break.

Would it? I thought that the old kernel should be enough to get upto pacman -Sy? Since once 7ji repo is added to pacman and the new kernel got installed by pacman, it shouldn't matter anymore.

kwankiu, 2023-09-03:
> So. After applying above fixed to AUR, amd arch-rock-installer it worked on my sd card.
>
> But i think using the old kernel would cause problems in the mid or long term because this kernel will not be [...]

I think that I will do something similar in the “custom kernel from repo” option that you can add custom repo url, add dkms and packages to install offline, etc.

but the standard installation method will be to use the old kernel that have network driver which should be less meant to break than always getting most up to date packages.

kwankiu, 2023-09-04:
@hbiyik the dkms packages now installed without error. But I noticed error when running systemctl start bluetooth.service

It should be a different error than the two missing 8852 firmware last time. 

I am not sure what’s the error yet.

hbiyik, 2023-09-05:
Error log? I dont gave that card so i cant reproduce.

kwankiu, 2023-09-05:
> Error log? I dont gave that card so i cant reproduce.

I really have to see … because this time i dont see any useful error log, dmesg looks normal, systemctl status bluetooth looks normal, bluetooth just doesnt turn on, i also tried to copying that two firmware file, same. Let me check the logs carefully and see.

closed 2023-09-30 by kwankiu