kwankiu/archlinux-installer #10: wifi only

ddams2000, 2023-09-23:
Hello, I wanted to know if it was possible to install your version of arch linux with wifi only as a connection?
cordially

kwankiu, 2023-09-27:
For now, not really.

You could try quitting the installer and manually set up internet, but I am not sure whether there are drivers and packages necessary to setup WiFi connection. (or you could try if anything like a USB WiFI or Thethering works).

I will see whether this is possible in the future. If yes, I will add support for WiFI for sure.

kwankiu, 2023-10-10:
Some status to this issue. So basically I have added experimental Wifi support to dev branch, but due to issues with the script setting up Arch Linux, things works randomly.

kwankiu, 2024-02-11:
WiFi support added in b04.

kwankiu, 2024-02-11:
This is only tested on RTL8852BE, what WiFi card are those? Do they work on Radxa’s official Debian image?

ddams2000, 2024-02-12:
> This is only tested on RTL8852BE, what WiFi card are those? Do they work on Radxa’s official Debian image?

it's a netgear a6210 usb wifi key (MT7612U) and the other is a noname pcie card (qca9377) neither works with the official distribution but they work with armbian

kwankiu, 2024-02-13:
I have no idea about what driver are required for your card to work, but here's what you can test with:

1. I just noticed that the linux-firmware-orangepi-git package comes with quite a lot of WiFi firmware, maybe you can try installing that.
2. If that package does not come with the WiFi firmware of your WiFi card, or (as i havent tested that package yet) if for any reason it doesnt work or breaks the install, you may need to find the required WiFi firmware package by yourself.

To install additional packages / firmware automatically, copy the package to /usr/lib/compiled-packages, you can either uncompress the image and copy the package, or at first boot, when it shows no internet and prompts with options, press Ctrl+C to terminate the installer, login with root/root and copy the package to /usr/lib/compiled-packages and then reboot

ddams2000, 2024-02-13:
Thank you for your help, I believe that my cards are not recognized at the kernel level, I tested a dmesg and there is no reference to my cards in the log :( (same on the official radxa image) :(

kwankiu, 2024-02-15:
Image is now built on linux-rockchip-joshua-git with linux-firmware-joshua-git (which is based on armbian firmware) to support more WiFi card.

You may want to try this image : 
https://github.com/kwankiu/archlinux-installer-rock5/releases/tag/b05-dev

ddams2000, 2024-02-15:
great it works... for the installation, once arrived on the plasma desktop I no longer have any card detected.

ddams2000, 2024-02-16:
Well done, it works perfectly! 
On the other hand I only tried kde and it's so unstable that it's unusable...

kwankiu, 2024-02-16:
What do you mean by unusable?

For KDE, idk why it sometimes freezes but u can simply restart it with :
killall plasmashell && kstart plasmashell

As far as I tested, gnome is the most stable.

ddams2000, 2024-02-16:
it really freezes all the time, when I open a terminal, I launch the browser, I click on the start menu maybe a problem with panfork.
Anyway I think this issue is resolved.
well done

closed 2024-02-16 by kwankiu