kwankiu/archlinux-installer #12: unable to boot

gcd0318, 2023-10-01:
I used the script to create a TF card, and insert to rock5b, but it did not boot when powered up. on-board LED is always green without blinking, and no display on screen.

I chose BSP mode and  TF is 16G, I'm not sure if I should provice any info more.

kwankiu, 2023-10-01:
Maybe try flashing this image instead.

https://github.com/kwankiu/archlinux-installer-rock5/releases/tag/b03

kwankiu, 2023-10-01:
> I used the script to create a TF card, and insert to rock5b, but it did not boot when powered up. on-board LED is always green without blinking, and no display on screen.
> 
> 
> 
> I chose BSP mode and  TF is 16G, I'm not sure if I should provice any info more.

If you are creating Arch Linux with the script, it only works in NVMe as for some reason the partition table created by parted doesn't boot with radxa u-boot if it is on a TF card. It somehow works perfectly on NVMe drive not TF Card. The best way for now is to use my pre-built image (or you can check the arguments used on the github workflow files, the workaround is to specify `-m -image=empty.img` which empty.img is a radxa's debian image that has all partition formatted (as an empty partition))

gcd0318, 2023-10-02:
cool, I'll try
Thanks a lot

closed 2023-10-02 by gcd0318