kwankiu/archlinux-installer #21: wlan0 vs nmtui

DarkevilPT, 2024-03-07:
Are you sure wlan0 is better than nmtui? For some reason I prefer nmtui and I wonder if nmtui isnt an better approach for this.. consider testing nmtui instead.. maybe..

kwankiu, 2024-03-07:
> But have u tought of testing it to see how the installer would behave in terms of process compared to re-asking which wifi to connect at each reboot during the installing?

Sure if I only have to support the rock 5 I will surely switch to nmtui

But I am pretty sure nmtui won’t work with Edge 2. If it works why would we be struggling fixing the WiFi issues.

kwankiu, 2024-03-07:
> > > But have u tought of testing it to see how the installer would behave in terms of process compared to re-asking which wifi to connect at each reboot during the installing?
> > 
> > 
> > Sure if I only have to support the rock 5 I will likely switch to nmtui
> > But I am pretty sure nmtui won’t work with Edge 2. If it works why would we be struggling fixing the WiFi issues.
> 
> It does works on edge 2. But only because the kernel is fixed for that now. Before connections couldnt be found (im not sure).

NOOOOOOOOOOOOOOOOOOOO

closed 2024-03-07 by kwankiu

kwankiu, 2024-03-25:
The idea of packing an online installation image is to keep the system clean and lightweight, nmtui needs a lot of depenency as well as full NetworkManager installed. The reason to use wifi-menu is because netctl came with arch linux itself, and wifi-menu only need wpa-supplicant and dialog to work.

netctl seems to support auto wifi connect, i have enabled it now, but needs some test