kwankiu/archlinux-installer #22: Installer root options

DarkevilPT, 2024-03-07:
When it asks what to do to the root user if disable remove it or keep it...

why bother with that step if we can do all that manually at the desktop environment and we already created an user by now?

Any chances u consider to skip this step to an turn if off automatically proceeding install? for a more smoother quicker process?

kwankiu, 2024-03-07:
This is part of the user creation, and some people may find it useful, there’s no reason to “remove a feature”.

and please note that this option is not available on ACU / Arch-Rock-Config and there is no GUI Settings from the Desktop Environment offer this feature (at least not gnome I believe)

closed 2024-03-07 by kwankiu

DarkevilPT, 2024-03-07:
Its as simples as going to users in settings and setting up an admin account. If no root then root should be visible there.

kwankiu, 2024-03-07:
That’s clearly a completely different thing.
The one in the installer is for user to choose what to do with the default root account.
I believe you may not be able to create a pure root account using any DE control center.

DarkevilPT, 2024-03-07:
Thats what I mean..dont waste time with that on the installer leaving that for the user to change on the DE by themselves.
But U dont agree so fine. We wont get an much click to go installer that way... with so manh prompts over installation rebooting and confirmations on languague... all those can even be set changed on the DE.. so I wouldnt care too much tbh. Armbian installer goes the same way when it comes to language but not to user root ( for that as I said ) users can edit post DE installation on user settings.