kwankiu/archlinux-installer #28: USB 2.0 doesn't work on Orange Pi 5

GongTae, 2024-03-09:
I was trying to connect a wireless mouse through port 2.0, and it receives no signal.

kwankiu, 2024-03-09:
Which kernel option did you choose during installation?
Is keyboard working through USB 2.0?

GongTae, 2024-03-09:
I chose the Joshua kernel, it works fine on USB 3.0.

kwankiu, 2024-03-09:
> I chose the Joshua kernel, it works fine on USB 3.0.

Does the same issue happens on Joshua’s Ubuntu image or does this issue only happen on Joshua’s kernel (like does this happen if you choose linux-aarch64-rockchip-orangepi ?)

If yes, you may want to open an issue at
https://github.com/Joshua-Riek/linux-rockchip

GongTae, 2024-03-10:
The problem does not exist when I use Ubuntu-Joshua, or when I use archlinux-7ji. (the problem with said USB2.0 port sharing the line with the USB-C port was present in the installer as well)

GongTae, 2024-03-10:
The problem still existing on release "b06"

GongTae, 2024-03-10:
To solve this, I used this file [https://github.com/Joshua-Riek/ubuntu-rockchip/blob/main/overlay/usr/lib/systemd/system/enable-usb2.service]
And I put it in the path: `/usr/lib/systemd/system/`
The next step was to activate the service with:
`sudo systemctl enable enable-usb2`
and start it with:
`sudo systemctl start enable-usb2`

closed 2024-03-10 by GongTae

kwankiu, 2024-03-10:
> To solve this, I used this file [https://github.com/Joshua-Riek/ubuntu-rockchip/blob/main/overlay/usr/lib/systemd/system/enable-usb2.service] And I put it in the path: `/usr/lib/systemd/system/` The [...]

Thank you for the information, I will add this fix soon.