kwankiu/archlinux-installer #33: error: linux-aarch64-rockchip-bsp6.1-joshua-panthor-git-headers: signature from "Samuel (kwankiu) <info@1us.ca>" is invalid

JFLim1, 2024-03-30:
Hi @kwankiu,

Tried to install 

linux-aarch64-rockchip-bsp6.1-joshua-panthor-git-6.1.43.r1266140.c30a5f70a982-1
linux-aarch64-rockchip-bsp6.1-joshua-panthor-git-6.1.43.r1266140.c30a5f70a982-1

But encountered error: 

Packages (2) linux-aarch64-rockchip-bsp6.1-joshua-panthor-git-6.1.43.r1266140.c30a5f70a982-1
             linux-aarch64-rockchip-bsp6.1-joshua-panthor-git-headers-6.1.43.r1266140.c30a5f70a982-1

Total Download Size:    10.80 MiB
Total Installed Size:  253.79 MiB

```
:: Proceed with installation? [Y/n] Y
:: Retrieving packages...
 linux-aarch64-rockchip-bsp6.1-joshua-panthor-git-headers-6.1.43.r1266140.c30a5f70 linux-aarch64-ro...    10.8 MiB  1957 KiB/s 00:06 [########################] 100%
(2/2) checking keys in keyring                     [########################] 100%
(2/2) checking package integrity                   [########################] 100%
error: linux-aarch64-rockchip-bsp6.1-joshua-panthor-git-headers: signature from "Samuel (kwankiu) <info@1us.ca>" is invalid
:: File /var/cache/pacman/pkg/linux-aarch64-rockchip-bsp6.1-joshua-panthor-git-headers-6.1.43.r1266140.c30a5f70a982-1-aarch64.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
Do you want to delete it? [Y/n] Y
error: failed to commit transaction (invalid or corrupted package (PGP signature))
Errors occurred, no packages were upgraded.
```

kwankiu, 2024-03-30:
With b07-dev?
I noticed this issue with the installer but I am not sure what’s causing this issue with pacman as the same configuration works on an existing system.

You can set SigLevel = Never to the repository 
Or just wait for me to rebuild the dev images

JFLim1, 2024-03-31:
> With b07-dev?

No.  Use it on an existing Archlinuxarm image following the prescribed instructions:

> Direct trust
> 
> import my signing key:
> 
> sudo pacman-key --recv-keys B669E3B56B3DC918
> sudo pacman-key --lsign B669E3B56B3DC918
> 
> add the following session in your /etc/pacman.conf:
> 
> [experimental]
> Server = https://github.com/kwankiu/archlinux-installer-rock5/releases/download/experimental

I suppose the headers is not required if just wanted to test Panthor.  Below installation command is sufficient get "panthor" working?

`sudo pacman -S linux-aarch64-rockchip-bsp6.1-joshua-panthor-git mesa-panfrost-git
`

kwankiu, 2024-03-31:
> No. Use it on an existing Archlinuxarm image following the prescribed instructions:
>
>
> I suppose the headers is not required if just wanted to test Panthor. Below installation command is sufficient [...]

Hmm, interesting because i had this working on existing system, you can set SigLevel= Never and it should work.

JFLim1, 2024-04-01:
> you can set SigLevel= Never and it should work

Hi @kwankiu ,
Thanks.  This did the trick to install "headers".

But when I tried to install "mesa-panfrost-git" encountered dependency error: **"mesa-panfork-git required by mpv-git"**
```

sudo pacman -S mesa-panfrost-git
resolving dependencies...
looking for conflicting packages...
:: mesa-panfrost-git-r185439.bfafaa944e3-1 and mesa-panfork-git-r164484.120202c6757-2 are in conflict (mesa-libgl). Remove mesa-panfork-git? [y/N] Y
error: failed to prepare transaction (could not satisfy dependencies)
:: removing mesa-panfork-git breaks dependency 'mesa-panfork-git' required by mpv-git
```

Does mpv-git works with "mesa-panfrost-git" with vpu hw acceleration? 

My device is Opi5-Plus.

kwankiu, 2024-04-01:
[image]
Umm I don’t see Mesa is dependency of mpv-git

@hbiyik any idea?

kwankiu, 2024-04-03:
@JFLim1 this repository will no longer be updated.
Please check out the new repository:
https://github.com/kwankiu/PKGBUILDs/releases/tag/experimental

closed 2024-04-03 by kwankiu