kwankiu/archlinux-installer #35: Gnome Minimal Broken

DarkevilPT, 2024-04-02:
[image]

Icons apps... useless... Broke big son.
I selected the mainline image 6.8.x

DarkevilPT, 2024-04-02:
[image]
Gnome Full after the reboot...

Also the last final prompt 'Installation complete you can reboot now (y/n)? doesnt even waits for me to type y and just reboots..

DarkevilPT, 2024-04-02:
https://github.com/kwankiu/archlinux-installer-rock5/assets/3583885/2bb0584a-95d1-4e9c-9e05-eae17d431a5a

Noice.

DarkevilPT, 2024-04-02:
didnt eray helped you ? fake sake mate... erayg? where is that big dev about? 😔 if I dont care you will keep it broken 😀 what will u tell me to do next? 'doesnt matter use gnome full' whats next ? kda plasma ? kek..

DarkevilPT, 2024-04-02:
[image]

kwankiu, 2024-04-02:
> didnt eray helped you ? fake sake mate... erayg? where is that big dev about? 😔 if I dont care you will keep it broken 😀 what will u tell me to do next? 'doesnt matter use gnome full' whats next ? kda plasma ? kek..

Nope ... Hmm, Idk what is changed in Gnome 46, so that this Arch Linux ARM updates broke Gnome completely for new installations

DarkevilPT, 2024-04-02:
Then this sucks.. we cant rely on the installer thanks to the fucking arch linux... only on unmantained images.. like BredOS... this is frustating and disapointing..

closed 2024-04-02 by DarkevilPT