kwankiu/archlinux-installer #61: Install to EMMC

mschirrmeister, 2024-08-21:
There is no way yet to install it from `sdcard` to `EMMC`? The wiki mentions a `--disk` option, but if I specify this I get an error.

Currently I flash directly to EMMC, which works fine. But was just wondering if there is or will be an option in `acu` for example to install to EMMC or NVME.

Working steps on macOS.

```
Install `rkdevtools`

    git clone https://github.com/radxa/rkdeveloptool.git
    cd rkdeveloptool
    autoreconf -i
    ./configure
    make

Boot device in `maskrom` mode, download the loader to the device and flash the image.

    curl -LO https://dl.radxa.com/rock5/sw/images/loader/rk3588_spl_loader_v1.15.113.bin
    
    sudo ./rkdeveloptool ld
    sudo ./rkdeveloptool db rk3588_spl_loader_v1.15.113.bin
    sudo ./rkdeveloptool wl 0 ~/Downloads/radxa-rock-5b-archlinux-installer.img
    sudo ./rkdeveloptool rd
```

kwankiu, 2024-09-06:
You can run this script to download and flash the installer from sd card to emmc.
This script automatically detects your device model but optionally, you can set `install_target=radxa-rock-5b`

```
curl https://raw.githubusercontent.com/kwankiu/archlinux-installer/dev/scripts/get-installer | install_disk=/dev/mmcblk0 sh
```

Note: Replace `/dev/mmcblk0` to the corresponding device you want to install on.

mschirrmeister, 2024-09-12:
Great, thank you. Tested it with a sd card. Will EMMC next rebuild a try.

closed 2024-09-12 by mschirrmeister