kwankiu/archlinux-installer #62: b08 - edge 2

DarkevilPT, 2024-09-03:
[image]

DarkevilPT, 2024-09-03:
[image]

DarkevilPT, 2024-09-03:
[image]

kwankiu, 2024-09-04:
The server seems overload/down, try again later or change the mirrorlist to a server available in your country.

Or use this to generate a mirrorlist available and optimized for your current internet.
```
# Add the misc repo (from source)
acu rem set misc https://github.com/1usOS/PKGBUILDs --branch=misc

# Install rate-mirrors
acu update && acu install misc/rate-mirrors

# Backup original pacman mirrorlist
sudo cp /etc/pacman.d/mirrorlist /etc/pacman.d/mirrorlist.bak

# Update your mirrorlist
rate-mirrors archarm | sudo tee /etc/pacman.d/mirrorlist
```

closed 2024-09-04 by kwankiu