kwankiu/archlinux-installer #65: Rock 5B+ Installer loops when trying to install to NVME

StephanieSunshine, 2024-11-08:
Greetings, I apologize ahead of time if this has been answered somewhere else. I tried to look at the closed issues before submitting this issue with no luck.

I have a Rock 5B+ that I went through the steps to enable NVME boot via the SPI bootloader update. Before, I tried to install to the NVME, I had used a USB stick with the same image (Latest for Rock5B+) and was able to get a booting instance after a few attempts. After that success, I decided to try and do the same, but from the NVME instead. I used gparted to zap all existing partition tables and then I dd'd the Arch for ARM image to the front of the NVME and rebooted. The installer came up and started the installation, after it rebooted a few times, initramfs fails because the root UUID changed. After manually mounting the correct partition from the NVME to /new_root and exiting, userland starts to boot, but gets to a point just after it syncs network time and reboots quickly without any clear error. I cannot escape this reboot loop.

Is there someway to disable this reboot loop from initramfs or from the working USB instance? I believe I have a mostly running system if I can just stop this and rebuild initramfs with the correct UUID for root and stop whatever is forcing this reboot.

StephanieSunshine, 2024-11-08:
I might have found a work around. From the initramfs shell, I mounted the nvme root to /new_root as before, this time I went in and edited /new_root/boot/extlinux/extlinux.conf to point from the old root=UUID=... to root=/dev/nvme0n1p2, saved, fsync'd, and rebooted. It seems part2 was able to restart its install from there, and after a little massaging of the Ethernet network device to enable, the part2 install would complete.

I can now reboot into a working system. One thing I did do, is in part2 I saw pacman -Syu have some package fails. I ctrl+c'd part2 at the very end before reboot, and reran this command by hand and then updated after it had completed.

StephanieSunshine, 2024-11-08:
Also to note (but the devs are probably already aware) I had to use the default recommended kernel and the 6.11 experimental would not boot.

mschirrmeister, 2024-12-07:
I think the 6.11 or now the 6.12 are not booting because there is no DTS file in the kernel source yet for this device. Armbian seems to have one in their 6.12 edge kernel. You could try to use their compiled dtb file and maybe it will work.

kwankiu, 2024-12-08:
> I think the 6.11 or now the 6.12 are not booting because there is no DTS file in the kernel source yet for this device. Armbian seems to have one in their 6.12 edge kernel. You could try to use their compiled dtb file and maybe it will work.

6.11/6.12 from linux-aarch64 or linux-aarch64-7ji doesn't have HDMI output for rock 5b, the dts is mainlined but with a vanilla mainline kernel there is no display output (so SSH only)

Joshua's 6.11 should work on rock 5b but his project is discontinued and I have never tested it, I might switch to Armbian's mainline kernel later which is a mainline kernel with patches for rk3588

kwankiu, 2024-12-08:
> I have a Rock 5B+ that I went through the steps to enable NVME boot via the SPI bootloader update. Before, I tried to install to the NVME, I had used a USB stick with the same image (Latest for [...]

Not sure if I understand what you mean, but for nvme to work, you should flash the u-boot image to the SPI according to Radxa's docs, then you would flash my image to the nvme directly, if you are moving an existing installation from another disk, especially you used any tools to copy or attempts to any partitions, the UUID might get changed so you would normally need to update your UUID in extlinux.conf to boot successfully.

mschirrmeister, 2024-12-08:
> 6.11/6.12 from linux-aarch64 or linux-aarch64-7ji doesn't have HDMI output for rock 5b, the dts is mainlined but with a vanilla mainline kernel there is no display output (so SSH only)
>
> Joshua's [...]

My use cases are only servers so HDMI is typically not important for me. For initial setup I use serial and then ssh.

Yes, the Armbian kernel is nice since it has many patches. We can just hope that more stuff gets sooner than later into the mainline kernel.

closed 2024-12-24 by kwankiu