kwankiu/archlinux-installer #67: Raspberry Pi 5 is support?

vincentaxhe, 2024-12-03:
Offically only raspberry 4B is offered prebuild image by Manjaro. But some peoplel also sucessfully use KDE on raspberry pi 5.
I download raspberrypi-archlinux-installer.img.xz and try use rpi-imager and usbimager to flash 'img' to udisk and nvme, all failed. can not login to arch menu, loop 'Unable to read partition as FAT', What I did wrong?

kwankiu, 2024-12-08:
> Offically only raspberry 4B is offered prebuild image by Manjaro. But some peoplel also sucessfully use KDE on raspberry pi 5. I download raspberrypi-archlinux-installer.img.xz and try use [...]

All Raspberry Pi 64-bit Models should work using the same image (they share the same bootloader/config/kernel), but my image is not well-tested for Raspberry Pi. I had a Pi 4B, and I was able to boot, but somehow, it never worked with my USB keyboard, so I have never been able to do any further testing. My 4B is currently used in my 3D Printer, so I am not able to test Arch on it.

vincentaxhe, 2024-12-09:
never mind, Endeavour Os offically offered rpi5 arm.

Motophan, 2025-02-14:
rpi 5 8gb 

For me keyboard works but I had a few issues w/ ethernet, wlan didnt work at all, it never loaded the desktop environment but I was able to get through. 

My goal is to see what latest kernel works if I can get it (not linux-rpi but linux-aarch64). Unfortunately this ends up wrecking my bootloader. This is the only project outside of official images I was able to actually boot.