kwankiu/archlinux-installer #70: Problems with Archlinux Installer to Nvme Disk on Orange PI Max

Hygens, 2024-12-23:
I attempt install on Orange PI Max but get that:
[image]

I follow instructions on:
https://github.com/kwankiu/archlinux-installer/wiki/Advanced

kwankiu, 2024-12-24:
You may want to look at my [GitHub workflows](https://github.com/kwankiu/archlinux-installer/blob/main/.github/workflows/build_image_uboot.yaml)

Currently, the preferred way to use this installer is to use my prebuilt image, you may want to [try](https://github.com/kwankiu/archlinux-installer/releases/tag/b08) that out instead.

kwankiu, 2024-12-24:
> ![image](https://private-user-images.githubusercontent.com/5247983/398236535-636d04cd-4752-4d38-ba25-4dbaef3844bc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUwNjk5MDcsIm5iZiI6MTczNTA2OTYwNywicGF0aCI6Ii81MjQ3OTgzLzM5ODIzNjUzNS02MzZkMDRjZC00NzUyLTRkMzgtYmEyNS00ZGJhZWYzODQ0YmMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjRUMTk0NjQ3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDJhYWQyODU1MGNkOTExN2VhNTgzNzdkYmZlODJjMjcxZDk4YzA3YmVhOTZhOTg4MzU5NGZlNzFjZmRiMzRlMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.iDZ6CC5XE6ppArn-78i5ZMuKU04InuPE7sG7P2Mrlss)

Btw i see that your url contains the installer's old repository name, github might stop redirecting it over time, so you would need to replace "archlinux-installer-rock5" with "archlinux-installer"

kwankiu, 2024-12-24:
> Btw i see that your url contains the installer's old repository name, github might stop redirecting it over time, so you would need to replace "archlinux-installer-rock5" with "archlinux-installer"

I just updated the URL in the wiki

but the advanced page wiki is currently WIP, it is recommended to rely on the github workflow and the `--help` output of the script

Hygens, 2024-12-24:
> You may want to look at my [GitHub workflows](https://github.com/kwankiu/archlinux-installer/blob/main/.github/workflows/build_image_uboot.yaml)
>
> Currently, the preferred way to use this installer [...]

Unfortunately that is the problem.
I flushed the image on one microSD 8GiB and attempted install on nvme after complete all steps  and errors because the system not permit download nothing to microSD after installed on microSD and for that not possible flush the nvme from microSD. That is the question!

kwankiu, 2024-12-24:
> I just updated the URL in the wiki
>
> but the advanced page wiki is currently WIP, it is recommended to rely on the github workflow and the `--help` output of the script

The following wiki is now updated
https://github.com/kwankiu/archlinux-installer/wiki/Advanced

kwankiu, 2024-12-24:
> Unfortunately that is the problem. I flushed the image on one microSD 8GiB and attempted install on nvme after complete all steps and errors because the system not permit download nothing to microSD [...]

Do you have a device with more than one nvme slot or a USB to nvme adapter? If yes, you can flash the image to the nvme the same way you flash the image to an sd card

If not, you might need to install it (or any other linux distro) to an sd card and then flash the image to the nvme on your SBC

Btw, a 8GB MicroSD card is unfortunately not sufficient to install a GUI Linux Distro, proceeding to install it would likely shows error, if you don't have a larger micro sd card, you can only install a cli (terminal) only system

Hygens, 2024-12-25:
Resolved with the same microSD 8GiB but with gnome minimal.
I have installed on microSD and after flush image to nvme direct on Orange PI 5 Max.
After that normal install.
I installed only to execute some tests and compare with other distros.

closed 2024-12-25 by Hygens