kwankiu/archlinux-installer #71: Install with kde minimal loops with invalid inputs

gnattu, 2025-01-01:
pacman is asking for a number (default=1) to select install candidates, but the installer script is spamming y which is invalid and the installation loops there and never stopped.

gnattu, 2025-01-01:
Just retried without any DE, and it is still prompting for install options.

kwankiu, 2025-01-18:
Does this issue persist? Which board and kernel are you using?

gnattu, 2025-01-19:
I believe this issue persists, likely because the `acu install -y` command fails to address the scenario where a package name has multiple providers. For instance, the packages `jack2` and `pipewire-jack` necessitate a selection. However, the acu responds with “y,” which is not a valid choice for pacman. Consequently, pacman prompts the question again, and the acu’s persistent “y” responses trigger this repetitive issue.

kwankiu, 2025-02-01:
Can you please tell me which board / image you are using?

Without that infomation it is hard to spot the issue, i have tested rk3588 and generic and it works without any issues, and for raspberry pi it is WIP.

gnattu, 2025-02-01:
It is Radxa Rock 5B flashed with raspberry pi imager repository

kwankiu, 2025-02-01:
> It is Radxa Rock 5B flashed with raspberry pi imager repository

That's weird, it works on my side, tested today (joshua 6.1 kernel, the recommended one)

Which kernel are you using?

Also, try b09-dev, before installing, select "edit mirrorlist", comment the default one and uncomment a few (3 or more) servers closest to your location

gnattu, 2025-02-01:
It is also joshua 6.1 kernel. Did yours really completed the stage2? Because acu -y does always spam y to pacman for packages with multiple choices. You never seen this?

kwankiu, 2025-02-01:
> It is also joshua 6.1 kernel. Did yours really completed the stage2? Because acu -y does always spam y to pacman for packages with multiple choices. You never seen this?

Yes, it does, it booted into gnome and is running fine right now.

gnattu, 2025-02-01:
Try with KDE?

kwankiu, 2025-02-01:
> Try with KDE?

Cant test with rk3588 right now, but I just tried installing KDE in a VM using my installer, and it works fine.

> or instance, the packages jack2 and pipewire-jack requires a selection when you install jack. However, the acu responds with “y,

If this is not an issue on pacman / the repo side, this could be an issue with my config in `kernel.yaml`, but this was fixed long time ago, 

```
        firmware:
          - linux-firmware-joshua-git
        graphics:
          - mesa
        video:
          - pipewire-jack
          - ffmpeg-mpp-git
        additional: null
```

In part 2, the kernel is installed and extlinux is configured, then it installs joshua firmware (which is basically network drivers), and mesa doesn't depend on jack as well, only ffmpeg does, but `pipewire-jack` should be installed before `ffmpeg`, unless `pipewire-jack` were not installed successfully, or your `kernel.yaml` is different, your described issue should not happen.

If the problem was `pipewire-jack` not installed successfully, it could be a network issue or the repo / mirror does not provide a valid package, switching to another mirror might help.

If the problem was `kernel.yaml` and the file / URL isn't modified, it could be that you are using an older copy of the file due to cache or an outdated mirror if you are using any proxy / VPN or any other provider/services to access a file from GitHub.

kwankiu, 2025-02-01:
You may also try removing `-y` ONLY in the video acceleration part : 

```
    ################################################################
    # Install Video Acceleration
    alert "Installing Video Acceleration"
    if [ -n "$install_video" ]; then
        acu -y install $install_video
    fi
```

You may use the file editor to edit `/usr/bin/installer` (to modify the installer's part2) before installing

gnattu, 2025-02-01:
Well it is easily reproducible just by `acu install -y plasma-meta`, you don't even need RK3588 for doing that, no?

kwankiu, 2025-02-01:
> Well it is easily reproducible just by `acu install -y plasma-meta`, you don't even need RK3588 for doing that, no?

I am aware that the `yes` command can not handle situations where pacman would ask for options. Therefore, I would avoid using package groups in video and graphics section, in the desktop environment section, I used `--noconfirm` instead of `-y` with my installer : 

```
    ################################################################
    # Install desktop environment
    alert "Installing Desktop environment"
    if [ -n "$install_desktop_environment" ]; then
        #acu -y install $install_desktop_environment
        acu install $install_desktop_environment --noconfirm
    fi
```

However, it is not ideal to use `--noconfirm` when handling non-upstream packages, for example `ffmpeg-mpp-git` would replace and conflict with `ffmpeg` or any other variants of this package, and `--noconfirm` would result in the said package not being installed.

closed 2025-02-15 by kwankiu