kwankiu/archlinux-installer #72: Bluetooth not Working on Gnome Minimal or Gnome Full

Hygens, 2025-01-09:
After some tests and attempt configure Bluetooth not work after install Arch Linux.
The result about systemctl status bluetooth is that:
<pre>
[jose@joseopi5max ~]$ systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
     Active: active (running) since Thu 2025-01-09 15:17:24 -03; 8min ago
 Invocation: cc986661266c47518b41b001268fc9a1
       Docs: man:bluetoothd(8)
   Main PID: 569 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 19126)
     Memory: 2.9M (peak: 3.4M)
        CPU: 52ms
     CGroup: /system.slice/bluetooth.service
             └─569 /usr/lib/bluetooth/bluetoothd

Jan 09 15:17:24 joseopi5max bluetoothd[569]: src/plugin.c:init_plugin() System does not support bap plugin
Jan 09 15:17:24 joseopi5max bluetoothd[569]: src/plugin.c:init_plugin() System does not support bass plugin
Jan 09 15:17:24 joseopi5max bluetoothd[569]: src/plugin.c:init_plugin() System does not support mcp plugin
Jan 09 15:17:24 joseopi5max bluetoothd[569]: src/plugin.c:init_plugin() System does not support vcp plugin
Jan 09 15:17:24 joseopi5max bluetoothd[569]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Jan 09 15:17:24 joseopi5max bluetoothd[569]: src/plugin.c:init_plugin() System does not support micp plugin
Jan 09 15:17:24 joseopi5max bluetoothd[569]: src/plugin.c:init_plugin() System does not support ccp plugin
Jan 09 15:17:24 joseopi5max bluetoothd[569]: src/plugin.c:init_plugin() System does not support csip plugin
Jan 09 15:17:24 joseopi5max bluetoothd[569]: src/plugin.c:init_plugin() System does not support asha plugin
Jan 09 15:17:24 joseopi5max bluetoothd[569]: Bluetooth management interface 1.22 initialized
</pre>

kwankiu, 2025-02-28:
AP6611S, currently the installer treats all orangepi5 model as AP6275P, seems like AP6611S needs a different command to enable bluetooth

Hygens, 2025-02-28:
> In `/usr/lib/systemd/system/bt_ap6275p.service`, replace the existing ExecStart with the one below:
> 
> ```
> #AP6611S Bluetooth
> ExecStart=/usr/bin/brcm_patchram_plus --bd_addr_rand --enable_hci --no2bytes --use_baudrate_for_download --tosleep 200000 --baudrate 1500000 --patchram /lib/firmware/SYN43711A0.hcd /dev/ttyS7
> ```

Thank you, worked now.

closed 2025-02-28 by Hygens