kwankiu/archlinux-installer #74: orangepi-5plus-archlinux-installer-uefi.img.xz

ageevp, 2025-01-26:
Stuck on a boot
But 
orangepi-5-plus-archlinux-installer.img.xz
works like a charm!!!!!!
Please check.

kwankiu, 2025-01-27:
Did you set to Device Tree Mode on first boot?

ageevp, 2025-01-28:
Yes!
I use vendor value.
The computer lights with lamps  like a chrismas tree
but  the screen is 
BLACK    !!!!!!!!!!!
Please help

kwankiu, 2025-02-01:
https://github.com/kwankiu/archlinux-installer/releases/download/UEFI-b09-dev/orangepi-5plus-archlinux-installer-uefi.img.xz

Can you see if this new build by any chance fixes the issue?

ageevp, 2025-02-03:
No. Without success!
Booting crash with no HDMI output.
Screen is black.

kwankiu, 2025-03-16:
@ageevp the new build now packs dtb to UEFI firmware from Joshua BSP6.1 to match with the kernel used among these rockchip images. So it should now boot.

However, due to dtb incompatitbility, when you choose any kernel other than Joshua BSP6.1, you will need to enable DTB overrides in UEFI settings and copy the kernel's dtb file (`/boot/dtbs/<kernel_package_name>/<board_platform>.dtb`) to [EDK2-RK3588's specify path](https://github.com/edk2-porting/edk2-rk3588?tab=readme-ov-file#custom-device-tree-blob-dtb-override-and-overlays) (`/boot/dtb/base/<platform>.dtb`)

ageevp, 2025-04-01:
Hi!
Now it boot!!!
I'm missing EDK2 requirements.
If configure as custom dtb and overlays it boot.
But do not include LCD overlay!!!!!

kwankiu, 2025-04-01:
> Hi! Now it boot!!! I'm missing EDK2 requirements. If configure as custom dtb and overlays it boot. But do not include LCD overlay!!!!!

Copy the lcd dtbo to
/boot/dtb/base/overlays/

Make sure the folder is `overlays` not `overlay`, and contains only the dtbo u want it to load.

ageevp, 2025-04-06:
Samuel!
 Make next version of image as livecd please!
With possibility of alongside installation.
With best regard Peter!

kwankiu, 2025-04-06:
> Samuel! Make next version of image as livecd please! With possibility of alongside installation. With best regard Peter!

https://github.com/kwankiu/archlinux-installer/issues/78#issuecomment-2688270482
This feature is on b09, would that helps instead?

ageevp, 2025-04-24:
No!
Installation onto partition do not work!

ageevp, 2025-04-24:
Could  not install multilib!

kwankiu, 2025-04-24:
> Could  not install multilib!
> 

multilib?

ageevp, 2025-04-25:
Yes. Include support multilib to next release.
And also GRUB.

kwankiu, 2025-05-21:
> Hi! Now it boot!!! I'm missing EDK2 requirements. If configure as custom dtb and overlays it boot. But do not include LCD overlay!!!!!

It is released to b09, closing.

Let me know to reopen this if there's still issues.

closed 2025-05-21 by kwankiu

ageevp, 2025-05-28:
Multilib is obsolet technology.
It've rejected.
llvm-mingw replace it.