kwankiu/archlinux-installer #75: Request

bobbyunknown, 2025-02-01:
Hallo can make for orange pi zero 3?

kwankiu, 2025-02-01:
Thanks for your interest to this project.

Currently, there are no support for allwinner based SBC with my installer, and I do not own an orange pi zero 3 or any h618 based SBC, so I might not be able to produce an image for orange pi zero 3 now.

However, it seems like this SBC is a popular low cost SBC for many projects and possibly an option for 3d printer, and i do have an allwinner h5 based nanopi, i might be able to add support to allwinner based SBC at a later time, but at this moment i cant tell if allwinner based SBC will be supported or not in the future.

bobbyunknown, 2025-02-18:
Thank bro
I have one more device rk3399 board rongpin king3399

kwankiu, 2025-02-25:
> rongpin king3399

Unfortunately this board doesnt seems to be supported by armbian or mainline.

closed 2026-09-02 by kwankiu