kwankiu/archlinux-installer #76: Providing Kernel image + dtbs + initramfs

peltho, 2025-02-02:
Hi, thank you for your work!
Is it somehow possible to provide the following files:
- kernel image
- initramfs image
- fdt / dtb files

I'm planning to serve them remotely to run Arch on distant SBC. At least how can I proceed to generate these files?
BR

kwankiu, 2025-02-02:
Can you be more specify on what you want to achieve and which SBC are you using? Kernel related files are available in the kernel package, and is usually installed under /boot, kernel packages are built from PKGBUILD depending on which kernel you need.

peltho, 2025-02-02:
I'm using rk3582 SoC (Rock 5C lite) and I want it to be netbooted.
I already have U-boot configured to fetch files (the ones mentioned earlier + rootfs) through pxe on a TFTP + NFS server.
Tell me if you need more info 🙏🏻