kwankiu/archlinux-installer #77: [rpi 5 issue] wlan and lan issues

Motophan, 2025-02-14:
This issue is actually a few independent issues and I can split these up if you need. 

1. Regardless of what I do (typing in wlan0 in adapter name) wlan will not work at all. 
2. If I load the generic linux kernel during stage 2 of the install after it reboots the LAN will not come up and restarting ethernet will not fix it
3. If I load either the rpi-linux kernel or the rpi-linux-16k kernel during stage 2 after it reboots the lan will not come up without restarting ethernet. It will come up after that. 

I am using the https://github.com/kwankiu/archlinux-installer/releases/tag/b08 release for 
raspberrypi-archlinux-installer.img.xz

TourioN, 2025-03-20:
I have the same problem on 4B

burntpiano, 2025-10-04:
issue still persists in b09 as well on my pi4b. ethernet is fine, but arch doesn't even see the wireless network interface regardless if it's integrated or usb.