kwankiu/archlinux-installer #82: [Request] Radxa Rock Pi 4B

danielclasen, 2025-03-04:
Hi @kwankiu 

any chance to get the Rock Pi 4B (or in that term the entire 4 series) supported? I am trying for myself to get at least the installer booted but no luck so far. 

```
./create-installation-media --model=radxa-rock-4b --kernel=linux-aarch64-rockchip-bsp6.1-joshua-git --addpkg=linux-firmware-joshua-git --bootfirmware=../rockchip-bsp/out/u-boot/spi
```

Where `../rockchip-bsp/out/u-boot/spi` is a build from the master at https://github.com/radxa/rockchip-bsp

I also tried with a mainline uboot build for rk3399 as described here: https://docs.u-boot.org/en/latest/board/rockchip/rockchip.html#building

I saw that you might be already working on Rock Pi 4 support and are looking for testers. I'd be more than happy to contribute in that regard!

Best,
Daniel

kwankiu, 2025-03-04:
https://github.com/kwankiu/archlinux-installer/releases/download/b09-dev/radxa-rock-pi-4b-archlinux-installer.img.xz

It is planned on next release, but i dont have the board to test

kwankiu, 2025-03-04:
> Hi [@kwankiu](https://github.com/kwankiu)
>
> any chance to get the Rock Pi 4B (or in that term the entire 4 series) supported? I am trying for myself to get at least the installer booted but no luck [...]

When u use the create media script, it is hard to tell without the log but the mainline u-boot might be incompatible with the vendor kernel, it is recommended to use radxa u-boot

danielclasen, 2025-03-04:
> https://github.com/kwankiu/archlinux-installer/releases/download/b09-dev/radxa-rock-pi-4b-archlinux-installer.img.xz
> 
> It is planned on next release, but i dont have the board to test

I flashed that image to my sdcard but it won't boot. I don't get any output on hdmi and don't have a serial console setup at hand right now.

kwankiu, 2025-03-04:
> I flashed that image to my sdcard but it won't boot. I don't get any output on hdmi and don't have a serial console setup at hand right now.

I added the trust firmware, does this fix it?

https://github.com/kwankiu/archlinux-installer/releases/download/b09-dev/radxa-rock-pi-4b-archlinux-installer.img.xz

danielclasen, 2025-03-04:
Sadly that still didn't do the trick. But I have hooked up a serial console and the full uboot log:

```
DDR Version 1.20 20190314
In
channel 0
CS = 0
MR0=0x98
MR4=0x3
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
CS = 1
MR0=0x18
MR4=0x1
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
channel 1
CS = 0
MR0=0x98
MR4=0x3
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
CS = 1
MR0=0x18
MR4=0x3
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
channel 0 training pass!
channel 1 training pass!
change freq to 400MHz 0,1
channel 0
CS = 0
MR0=0x98
MR4=0x81
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
CS = 1
MR0=0x18
MR4=0x1
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
channel 1
CS = 0
MR0=0x98
MR4=0x3
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
CS = 1
MR0=0x18
MR4=0x81
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
channel 0 training pass!
channel 1 training pass!
change freq to 800MHz 1,0
Channel 0: LPDDR4,800MHz
Bus Width=32 Col=10 Bank=8 Row=15/15 CS=2 Die Bus-Width=16 Size=2048MB
Channel 1: LPDDR4,800MHz
Bus Width=32 Col=10 Bank=8 Row=15/15 CS=2 Die Bus-Width=16 Size=2048MB
256B stride
ch 0 ddrconfig = 0x101, ddrsize = 0x2020
ch 1 ddrconfig = 0x101, ddrsize = 0x2020
pmugrf_os_reg[2] = 0x3AA1FAA1, stride = 0xD
OUT
Boot1: 2018-06-26, version: 1.14
CPUId = 0x0
ChipType = 0x10, 221
Spi_ChipId = b4018
SpiBootInit:0
mmc0:cmd5,32
SdmmcInit=0 0
BootCapSize=0
UserCapSize=60906MB
FwPartOffset=2000 , 0
StorageInit ok = 200234
SecureMode = 0
SecureInit ret = 0, SecureMode = 0
GPT  signature is wrong
LoadTrust Addr:0x1800
LoadTrust Addr:0x1c00
LoadTrust Addr:0x2000
No find bl30.bin
No find bl32.bin
Load uboot, ReadLba = 1000
hdr 00000000031912c0 + 0x0:0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,

tag:LOADER error,addr:0x3000
hdr 00000000031912c0 + 0x0:0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,

Load uboot, ReadLba = 1400
hdr 00000000031912c0 + 0x0:0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,

tag:LOADER error,addr:0x3400
hdr 00000000031912c0 + 0x0:0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,

Load uboot, ReadLba = 1800
hdr 00000000031912c0 + 0x0:0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,

tag:LOADER error,addr:0x3800
hdr 00000000031912c0 + 0x0:0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,

Load uboot, ReadLba = 1c00
hdr 00000000031912c0 + 0x0:0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,

tag:LOADER error,addr:0x3c00
hdr 00000000031912c0 + 0x0:0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,

Load uboot, ReadLba = 2000
Load OK, addr=0x200000, size=0xfca20
RunBL31 0x40000
NOTICE:  BL31: v1.3(release):845ee93
NOTICE:  BL31: Built : 15:51:11, Jul 22 2020
NOTICE:  BL31: Rockchip release version: v1.1
INFO:    GICv3 with legacy support detected. ARM GICV3 driver initialized in EL3
INFO:    Using opteed sec cpu_context!
INFO:    boot cpu mask: 0
INFO:    If lpddr4 need support multi frequency,
INFO:    please update loader!
INFO:    Current ctl index[0] freq=400MHz
INFO:    Current ctl index[1] freq=800MHz
INFO:    plat_rockchip_pmu_init(1196): pd status 3e
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot latest-2023.10-9-eed05a18-geed05a18 (Jan 30 2024 - 07:55:02 +0000)

SoC: Rockchip rk3399
Reset cause: POR
Model: Radxa ROCK 4B
DRAM:  4 GiB (effective 3.9 GiB)
Device tree error at node '__symbols__'
Some drivers failed to bind
initcall sequence 00000000f7fc5180 failed at call 00000000002257b4 (err=-11)
### ERROR ### Please RESET the board ###
```

danielclasen, 2025-03-04:
Hmm don't know if I don't get it or if there is an actual issue with the build. The extlinux.conf looks like this (both, for my local builds as well for the img you provided):

```
# This extlinux.conf is auto-generated by arch-installer
 
label linux
    kernel /Image
    initrd /boot/initramfs-linux.img
    devicetreedir /boot/dtbs/linux
    append   root=UUID=e9cac016-ec94-40ab-a943-335fe2188faf console=ttyFIQ0,1500000n8 quiet splash loglevel=4 rw earlycon consoleblank=0 console=tty1 coherent_pool=2M irqchip.gicv3_pseudo_nmi=0 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
```

But the `devicetreedir /boot/dtbs/linux` does not exist. Should that maybe be set to `devicetreedir /boot/dtbs/rockchip` instead? 

I know that it is sed explicitly here https://github.com/kwankiu/archlinux-installer/blob/b9712cbaf4164176604aecb220ab543e90a5f103/create-installation-media#L1044 but when building mainline $argkrl will always be `linux` or using the kernel from the matrix build it would be `linux-aarch64-rockchip-bsp6.1-joshua-git` while the actual dtb resides in `/boot/dtbs/rockchip/rk3399-rock-pi-4b.dtb`.

danielclasen, 2025-03-05:
Okay, now we're cookin' with gas. So I grabbed u-boot from here https://forum.radxa.com/t/rock-pi-4b-debian-kde-not-boot-from-nvme/19625/21 and did build without a special kernel like this:

```
./create-installation-media --model=radxa-rock-pi-4b --bootfirmware=usr/lib/u-boot-fix
```

After that I had to fix the `kernel` path and replaced `devicetreedir` with `fdt` to point to the dtb file directly. So I changed the `extlinux.conf` to:

```
# This extlinux.conf is auto-generated by arch-installer
 
label linux
    kernel /boot/Image
    initrd /boot/initramfs-linux.img
    fdt /boot/dtbs/rockchip/rk3399-rock-pi-4b.dtb
    append   root=UUID=e9cac016-ec94-40ab-a943-335fe2188faf console=ttyFIQ0,1500000n8 quiet splash loglevel=4 rw earlycon consoleblank=0 console=tty1 coherent_pool=2M irqchip.gicv3_pseudo_nmi=0 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
```

Before it looked like in my previous reply. 

Without `kernel` path being not properly prefixed with `/boot` uboot will just try to fallback to netboot stuff.

So with those fixes applied I am able to boot the installer 🎉 

```
DDR Version 1.20 20190314
In
soft reset
SRX
channel 0
CS = 0
MR0=0x98
MR4=0x1
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
CS = 1
MR0=0x18
MR4=0x1
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
channel 1
CS = 0
MR0=0x98
MR4=0x1
MR5=0xFF
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
CS = 1
MR0=0x18
MR4=0x3
MR5=0xFF
MR8=0x8
MR12=0x72
[... 165 lines cut ...]
scanning bus usb@fe380000 for devices... 1 USB Device(s) found
scanning bus usb@fe3c0000 for devices... 1 USB Device(s) found
scanning bus usb@fe800000 for devices... 5 USB Device(s) found
scanning bus usb@fe900000 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
rockchip_pcie pcie@f8000000: PCIe link training gen1 timeout!
Hit Ctrl+C key in 0 seconds to stop autoboot...
** Booting bootflow 'mmc@fe320000.bootdev.part_2' with extlinux
1:      linux
Retrieving file: /boot/Image
Retrieving file: /boot/initramfs-linux.img
append: root=UUID=c29dc4ca-6f62-4671-b615-7bd3f4ba0122 console=ttyFIQ0,1500000n8 quiet splash loglevel=4 rw earlycon consoleblank=0 console=tty1 coherent_pool=2M irqchip.gicv3_pseudo_nmi=0 cgroup_enable=cpuset 
cgroup_memory=1 cgroup_enable=memory swapaccount=1
Retrieving file: /boot/dtbs/rockchip/rk3399-rock-pi-4b.dtb
Moving Image from 0x2080000 to 0x2200000, end=4bd0000
## Flattened Device Tree blob at 01f00000
   Booting using the fdt blob at 0x1f00000
Working FDT set to 1f00000
Host not halted after 16000 microseconds.
   Loading Ramdisk to f479b000, end f4ed15e8 ... OK
   Loading Device Tree to 00000000f4784000, end 00000000f479a7eb ... OK
Working FDT set to f4784000

Starting kernel ...
```

danielclasen, 2025-03-05:
Maybe u-boot mainline would also be an option to try, at least that's what is being used in manjaro-arm: https://gitlab.manjaro.org/manjaro-arm/packages/core/uboot-rockpi4b/-/blob/master/PKGBUILD?ref_type=heads ultimately grabbing from here: https://source.denx.de/u-boot/u-boot/-/tree/v2023.01/board/rockchip/evb_rk3399

But seems to be updated also in the most recent release of u-boot:
https://source.denx.de/u-boot/u-boot/-/tree/v2025.01/board/rockchip/evb_rk3399?ref_type=tags and https://source.denx.de/u-boot/u-boot/-/tree/v2025.01/board/radxa/rockpi4-rk3399?ref_type=tags

kwankiu, 2025-03-07:
> Sadly that still didn't do the trick. But I have hooked up a serial console and the full uboot log:
> 
> U-Boot latest-2023.10-9-eed05a18-geed05a18 (Jan 30 2024 - 07:55:02 +0000)
> 
> SoC: Rockchip rk3399
> Reset cause: POR
> Model: Radxa ROCK 4B
> DRAM:  4 GiB (effective 3.9 GiB)
> Device tree error at node '__symbols__'

device tree error, so that the u-boot probably does not define fdt file path

in that case, devicetreedir can not be used

when using `linux-aarch64-rockchip-bsp6.1-joshua-git`

The fdt path should be `/boot/dtbs/linux-aarch64-rockchip-bsp6.1-joshua-git/rockchip/rk3399-rock-pi-4b.dtb`
devicetreedir would be /boot/dtbs/linux-aarch64-rockchip-bsp6.1-joshua-git as u-boot sets fdt file path as `rockchip/<platform>.dtb`, but in this case, fdt file path was probably not yet implemented on rk3399 radxa boards

dclasen-qd, 2025-03-07:
Yep, I am preparing a PR over here: https://github.com/kwankiu/archlinux-installer/compare/dev...dclasen-qd:archlinux-installer:dev

kwankiu, 2025-03-07:
> Yep, I am preparing a PR over here: [dev...dclasen-qd:archlinux-installer:dev](https://github.com/kwankiu/archlinux-installer/compare/dev...dclasen-qd:archlinux-installer:dev)

Thanks

While boot.yaml support using exact dtb path but having one boot config per board variant might not be ideal as this would creates a long list where the only difference between them is just the platform name in dtb.

kwankiu, 2025-03-07:
> Maybe u-boot mainline would also be an option to try, at least that's what is being used in manjaro-arm: [...]

The mainline u-boot does seems to have a default fdt file path
https://github.com/u-boot/u-boot/blob/0fd7ee0306a88bd99d3820167fa45a7ee6fbc1e1/configs/rock-pi-4-rk3399_defconfig#L25

But is all rock pi 4 variants mainlined to u-boot?

dclasen-qd, 2025-03-07:
> Thanks
>
> While boot.yaml support using exact dtb path but having one boot config per board variant might not be ideal as this would creates a long list where the only difference between them is just [...]

Yeah that's ugly right now, also not a fan of it :D I am trying to get it working with mainline alarm and u-boot right now. Might have to provide custom builds for the Rock Pi 4 Series as you discovered though. Currently only the Rock Pi 4a and 4c seem to be mainlined in u-boot. After that I am going to pretty up my changes.

However with the (currently ugly) config hacks, I was able to install a minimal alarm system with linux-aarch64 and the u-boot I found over at the radxa forum. Something in `acu` was weird though. I got this error when the `linux-aarch64` kernel got updated through the installer: https://github.com/kwankiu/acu/blob/4d2a383d2b1407eea32999e5f2e52d3267e1291b/acu#L1587