kwankiu/archlinux-installer #84: [feedback] Radxa ROCK 5C Lite

c0m4r, 2025-03-15:
Hi, first I wanted to say i really like your installer, great job and I also good sense of humor :)
I would like to share my experience with the installer on a Radxa ROCK 5C (Lite).
The installer works quite well and eventually installs the system, but there are a few issues. I thought my feedback might be helpful. If you need any additional information or re-testing on my end, I'd be happy to help.
I tried both [b08](https://github.com/kwankiu/archlinux-installer/releases/tag/b08) and [b09-dev](https://github.com/kwankiu/archlinux-installer/releases/tag/b09-dev).
I tried installing on two independent LANs and on two different SD cards to rule out problems of this nature.
Lastly, just to clarify, I use Arch on a different device on a daily basis, and the problems I encountered are specific to this device, which I just got. And of course I've already tried an official image from Radxa based on Debian and also Armbian - I didn't encounter similar problems on them.

#### 1) sudo and glibc

There is definitely a problem with installing sudo and glibc and with libs in general. I saw that the installer tries to install them, the progress bar from pacman appears, but after a few seconds it disappears and the installation continues, however after restarting sudo it is not installed and the libraries are missing. Maybe it has something to do with the type of network connection, because I didn't use WIFI. I was connected with an ethernet cable. I haven't seen any problems with the network. The computer gets an address from DHCP, DNS is set, everything looks OK network-wise.

```bash
[root@radxa ~]# htop
htop: /usr/lib/libc.so.6: version `GLIBC_2.38' not found (required by htop)
[root@radxa ~]# sensors
sensors: /usr/lib/libc.so.6: version `GLIBC_2.38' not found (required by /usr/lib/libsensors.so.5)
[root@radxa ~]# ldconfig 
[root@radxa ~]# sensors
sensors: /usr/lib/libc.so.6: version `GLIBC_2.38' not found (required by /usr/lib/libsensors.so.5)
[root@radxa ~]# btop
btop: /usr/lib/libc.so.6: version `GLIBC_2.38' not found (required by btop)
btop: /usr/lib/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by btop)
btop: /usr/lib/libstdc++.so.6: version `CXXABI_1.3.15' not found (required by btop)
```

I also had the installer flicker multiple times during the installation with these messages appearing shortly:

```bash
/usr/bin/acu: line 61: sudo: command not found
/usr/bin/acu: line 61: sudo: command not found
/usr/bin/acu: line 61: sudo: command not found
/usr/bin/acu: line 61: sudo: command not found
```

Additionally, when trying to reinstall sudo in one case it failed and the process hung, after which sudo did not work. In the second case pacman finished reinstalling, but sudo still did not work. I had to manually remove all remnants of it and install it fresh.

Finally, when I managed to solve the problems with sudo and glibc and most programs started working, suddenly I had issues with pacman, which suddenly was unable to read openssl libraries even though other programs like curl had no problem whatsoever. I had no more patience, so I didn't dig any deeper :)

#### 2) Operation too slow

Another strange problem is an error that appears at random when trying to install a package from the repo.

```bash
[root@radxa ~]# pacman -S glib2
resolving dependencies...
looking for conflicting packages...

Packages (1) glib2-2.84.0-1

Total Download Size:    4.66 MiB
Total Installed Size:  40.84 MiB
Net Upgrade Size:      15.02 MiB

:: Proceed with installation? [Y/n] 
:: Retrieving packages...
 glib2-2.84.0-1-aarch64.pkg.tar.xz failed to download
error: failed retrieving file 'glib2-2.84.0-1-aarch64.pkg.tar.xz' from mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
warning: failed to retrieve some files
error: failed to commit transaction (download library error)
Errors occurred, no packages were upgraded.
```

I had to run the command several times until it finally works.

#### 3) fan control

Auto pwm (step_wise) doesn't work, the fan runs at max all the time (ofc I can change its speed manually and workaround it with a script, so it's not a big deal)

kwankiu, 2025-03-15:
Well, if it is not operation too slow (mirror issue) but sudo failed to install even with ethernet, that’s a bit werid

For the too slow error, (for b09/b09-dev) before you install, go to network options -> edit mirrorlist, comment the default (first) mirror URL, and enable at least 2-3 closest mirror to your location 

And for rock 5d, no i dont mean rock 5t, i believe radxa has a board with code name rock 5d that is probably not yet announced

kwankiu, 2025-03-16:
I would actually assume 1) and 2) are probably the same issue (alarm mirror issue), which is an issue with Arch Linux ARM's own server, as i mentioned eariler, for now you can manually comment the default server and uncomment a few of them that works best for you.

I am planning on making this part of the installation so that user wont miss this step out and get server stability / availability issues.

c0m4r, 2025-03-16:
Hmm, I didn't look at it that way, but indeed 1 and 2 may be related. I rarely change the default mirror because everything works, but indeed the infrastructure of the arm arch linux version is limited so this may be a problem. I'll try testing this again later. Thanks!

Interesting thing about this 5D, maybe they are actually cooking something new :)

c0m4r, 2025-03-17:
It turns out that the problems are not related. Changing the mirror only solved problem 2 "Operation too slow". However, the installer still had a problem with the installation, but that's because ethernet doesn't start (that is, at some stage it starts, but later, after a restart I had to manually raise the interface and manually configure it via ip or activate dhcpcd).

```
ip link set end1 up
systemctl enable dhcpcd --now
```

What is extremely strange is that the installer was able to install glibc, but not the sudo, it's like it had connection for a moment but lost it, but I didn't find any obvious indication of that in the logs.

In any case, if you are connected via ethernet, before running the installation you should exit to the shell, change the mirror, bring up the interface, configure the network manually or via dhcp and the installation should be successful.

Some suggestions to consider:

- you might want to consider the `--needed` option in pacman commands, this will save time when the installer has to repeat the same steps again
- imho this should be more aparent that the installer will use custom repo (7Ji), there is no information about it anywhere in the readme, nor wiki. I guess it's hard to avoid this, due to the dispersion of development of different components and the long process to mainline, but even more so from a security point of view it should be transparent where the software that is installed comes from
- even though I chose to install without a graphical environment, the installer installs packages such as mesa, alsa and other graphical tools that are "redundant" for a server/headless
- when using "exit to shell" I don't actually go to the shell, only to log in, so it would be nice if the prompt showed information about the default login and password (I guessed it, but someone else might be surprised)

kwankiu, 2025-04-07:
> * you might want to consider the `--needed` option in pacman commands, this will save time when the installer has to repeat the same steps again

This should only be an issue with installing desktop environment right now, there is already a check before installing sudo instead of using `--needed`, but thanks for the suggestion.

> imho this should be more aparent that the installer will use custom repo (7Ji), there is no information about it anywhere in the readme, nor wiki. I guess it's hard to avoid this, due to the [...]

When u pick a kernel option it states which repo it comes from, custom repo is only added when u choose that kernel, but thanks i will mention this in wiki, on b09-dev i have also added a misc repo as default, this provides patched packages on top of Arch Linux ARM (for example alarm's grub is not compatitble with its own linux-aarch64 kernel, but alarm doesnt wants to fix it) as well as some useful AUR packages for alarm.

kwankiu, 2025-04-07:
> when using "exit to shell" I don't actually go to the shell, only to log in, so it would be nice if the prompt showed information about the default login and password (I guessed it, but someone else might be surprised)

[image]

kwankiu, 2025-05-21:
I am closing this.

For fan control, maybe report this to the kernel maintainer. Are you using Joshua 6.1 or Armbian 6.1? or 5.10?

closed 2025-05-21 by kwankiu