kwankiu/archlinux-installer #85: pacman doesn't retry when some mirrors fail

Popolon, 2025-04-06:
During the installation process, if a mirror is too slow, then some failure occurs. The script don't catch the error and continue to the next steps. Some important packages can so be missing.

A solution could be to wrap pacman in a function catch a return code, and retry until rc=0

```bash
function pacman_call {
 rc=1
 while [ rc != 0 ]
  pacman -S "$@"
  rc=?$
}

pacman_call base-devel glibc
```

kwankiu, 2025-04-06:
> During the installation process, if a mirror is too slow, then some failure occurs. The script don't catch the error and continue to the next steps. Some important packages can so be missing.
> 
> A solution could be to wrap pacman in a function catch a return code, and retry until rc=0
> 
> function pacman_call {
>  rc=1
>  while [ rc != 0 ]
>   pacman -S "$@"
>   rc=?$
> }
> 
> pacman_call base-devel glibc

That keeps trying until it succeed, but when the mirror are too slow (unreliable), this potentially traps the installation into an infinite loop, even if it somehow works after countless retries, the time taken to install and the huge amount of server request isnt ideal.

I believe that the current installer would actually retries once (acu install -y fallback to interactive once if failed) when pacman returns failure (to my understanding you are trying to catch any non-successful return status, which does not capture specifically the mirror too slow error, but captures any error)

The main issue is that Arch Linux ARM's official mirror is very unreliable but the mirrorlist has only one mirror enabled, and some mirrors on their list are broken. On b09, my `edit mirrorlist` feature downloads an updated mirrorlist, comment out the first (default) mirror and uncommenting at least 2-3 mirrors nearest to your location should works well.

kwankiu, 2025-05-21:
b09 now has an option to manage mirrors, the recommended / default options enable and rank fastest mirror according to your internet.

I will close this.

closed 2025-05-21 by kwankiu